RogueMaster/flipperzero-firmware-wPlugins #329: Possible Magspoof Module ?

vivinano, 2022-09-17:
### Describe the enhancement you're suggesting.

I saw there was a magspoof module in an old version of flipper is there a way to convert that for the newer firmware 

https://github.com/dunaevai135/flipperzero-firmware/tree/feature/nomodule/applications/magspoof

if not able to just let me know

RogueMaster, 2022-09-17:
Don’t have the hardware or time, sorry

closed 2022-09-17 by RogueMaster

0ki, 2022-09-21:
Hold up, what do you mean, hardware? It should work with the built in LF RFID antenna.

Can this module be compiled in or are there format changes that need to be made?

RogueMaster, 2022-09-21:
Changes are needed for sure... this one may be closer... https://github.com/wetox-team/flipperzero-firmware/tree/ezhevita/magspoof, if you compile it and get it to work we can add it in for others to try too

RogueMaster, 2022-09-21:
Also, if using internal antenna, we need some sort of guarantee or clarification that it wont brick RFID after continuous use, otherwise, external antenna is better