RogueMaster/flipperzero-firmware-wPlugins #333: Allow signal replay directly from the SubGhz Read screen

cusspvz, 2022-09-20:
### Description of the feature you're suggesting.

Adding the ability to read Raw signals directly from the Read screen was a great idea. Although, testing if the signal was correctly decoded by Flipper you got to first save it, then loose all the other signals that you might have captured and then “send”/replay the signal.

I wonder how easy would it be to add up the send action to the signal info screen directly.

### Anything else?

I've asked at the flipper OFW repo (flipperdevices/flipperzero-firmware#1732) but it seems like this issue is specific to this firmware version.

RogueMaster, 2022-09-21:
OFW does not have Read Raw from Read screen.
It is known to not work to replay the items read unless saved first.
Adding the option to Send would make it crash since it doesn't Send from memory too well.

Thoughts @perspecdev ?

perspecdev, 2022-09-21:
It's definitely possible to write code to allow them to be sent that way, but it's not set up for the raw protocol. I probably don't have time to set it up, but it's possible to do. If I remember correctly, there's a decoder function that needs to be filled out in raw.c

RogueMaster, 2022-09-21:
Hi @cusspvz I see the value in your request so I'll leave this open but I am not sure on how this will need to be fixed upon. Feel free to PR if you find a solution. Perhaps you can recruit someone on Discord to help fix the issue

RogueMaster, 2022-10-02:
Please check if your problem is fixed with https://github.com/RogueMaster/flipperzero-firmware-wPlugins/releases/latest

closed 2022-10-05 by RogueMaster