RogueMaster/flipperzero-firmware-wPlugins #337: Soft lock on power timeout while connected to USB and using an app

gregdan3, 2022-09-21:
### Describe the bug.

The Flipper Zero gets stuck on the "It is now safe to disconnect USB" screen until you force a restart, under the listed reproduction conditions.

### Reproduction

1. Connect Flipper Zero via USB
2. Enable power timeout in settings
3. Enter Frequency Analyzer
4. Leave running until power timeout occurs
5. Follow on screen instructions: Disconnect USB
6. Observe the screen still on (and the LED still flashing)

### Logs

```Text
No logs available.
```

RogueMaster, 2022-09-21:
Yes, for flipper to shut down it requires that it be unplugged... not sure why that's just how it is... makes little sense since it restarts for DFU install just fine without the unplug...

closed 2022-09-21 by RogueMaster

RogueMaster, 2022-09-21:
Please do mention your findings on [#1647](https://github.com/flipperdevices/flipperzero-firmware/pull/1647)