RogueMaster/flipperzero-firmware-wPlugins #342: Add console logging to marauder companion

pacmandu, 2022-09-23:
### Description of the feature you're suggesting.

Hi there! I was wondering if console logging to a text file on the flipper could be added to the Wifi Marauder Companion plugin? That way any data data captured like the PMKID is saved and full logs are backed up to the SD card without needing to solder to the dev board.

RogueMaster, 2022-09-23:
This isn't the site for the Wifi Marauder Companion plugin, please inquire with the author

pacmandu, 2022-09-23:
Understood. Where is the official repo for this plug-in? It's not in the official Marauder repo.

RogueMaster, 2022-09-23:
https://github.com/0xchocolate/flipperzero-firmware-with-wifi-marauder-companion