RogueMaster/flipperzero-firmware-wPlugins #353: Provide .tgz firmware package

augustozanellato, 2022-10-03:
### Description of the feature you're suggesting.

Providing a .tgz firmware package (similar to the one Unleashed provides) would make it possible to flash/update RM firmware using mobile app

RogueMaster, 2022-10-03:
Sure, what command set is needed for the folder structure and compression for it to work?

augustozanellato, 2022-10-03:
> Sure, what command set is needed for the folder structure and compression for it to work?

Check here https://github.com/flipperdevices/flipperzero-firmware/blob/0f9ea925d30f7a6183abe1002d680e3d1613a159/.github/workflows/build.yml#L84, should be pretty easy.

RogueMaster, 2022-10-03:
Sorry, that command doesn't yield the needed result... after running:

root@debian:~/fap/flipperzero-firmware-wPlugins# ./fbt updater_package

What is needed?

RogueMaster, 2022-10-04:
Ok, will try on this release

RogueMaster, 2022-10-04:
Have you tried installing the resulting file? Doesn't seem to work on qFlipper, does it work on mobile?

closed 2022-10-04 by RogueMaster