RogueMaster/flipperzero-firmware-wPlugins #356: SD Card Home Screen Animations lag Flipper Zero

EpicLPer, 2022-10-05:
### Describe the bug.

The new high FPS/smooth animations which (presumably) are loaded from the Micro SD are lagging the Flipper Zero immensely at times when it can't read the animation fast enough, sometimes hanging for over 5 seconds before responding again, this also occurs while in the menu trying to scroll through options.

### Reproduction

1. Turn Flipper On
2. Wait till Home Screen animation loads (can take ~5 seconds)
3. Let it play for a few
4. Animation and Flipper both start freezing, including Menu Scrolling sometimes also freezing with it

gztproject, 2022-10-05:
I experienced the same issue and simply deleted most of the extra animations. This speeds up the Flipper significantly.

RogueMaster, 2022-10-05:
Thanks, I'll try to give animations more resources to fix this.... this is a ticket nobody needed... please submit a PR if you find solutions...

closed 2022-10-05 by RogueMaster

EpicLPer, 2022-10-05:
The solution is to not include as resourceful animations? No need for a rather rude answer like that...