RogueMaster/flipperzero-firmware-wPlugins #406: 0.70.2-1104-RM1839 firmware install failed

4DA, 2022-11-05:
### Describe the bug.

I am trying to install 0.70.2-1104-RM1839 firmware on top of vanilla flipper 0.70.1 and at the end see "Update failed" message, "Error 13-0"

[image]
After this failed update I can see updated desktop but no application is able to start, showing errors.

### Reproduction

1. Vanilla flipper firmware 0.70.1
2. Extract tgz file to flipper sd card (includuing sub resources catalogue)
3. From flipper browser run updater app

RogueMaster, 2022-11-05:
try again?

RogueMaster, 2022-11-05:
I don't get this error

closed 2022-11-05 by RogueMaster

4DA, 2022-11-06:
okay, my isssue what that I uncompressed resources.tar and deleted the archive (bc read in some other fw to do it) :D

RogueMaster, 2022-11-06:
yeah, that's wrong, for all firmwares...