RogueMaster/flipperzero-firmware-wPlugins #407: Keyboard layout is wrong

XuliGan4eg2006, 2022-11-06:
### Describe the bug.

Hello! On this custom FW I have issue with bad usb (at the same time in official FW it has fix). Bad usb starts typing commands and ignores keyboard language. (I have changed layout but no result)

### Reproduction

1. Switch keyboard layout on your pc to any except En
2. Start any Bad usb script

### Anything else?

https://imgur.com/a/aiCrPEU - video demonstration
My config: 
[image]
[image]
(newest FW)

ESurge, 2022-11-09:
I don't seem to have this issue.

Tested changing Keyboard layout to Portuguese (`pt-PT`) with `pt-PT` in BadUSB, and Keyboard layout to French (`fr-FR`) with `fr-FR` in BadUSB.

I think the issue is that the BadUSB layouts doesn't have Russian `ru-RU` as a supported language.

ESurge, 2022-11-09:
A workaround for this is to use AltString in the BadUSB script.

closed 2022-11-09 by RogueMaster