RogueMaster/flipperzero-firmware-wPlugins #427: Error number(1-60)

Yourit7870, 2022-11-20:
### Describe the bug.

"Update Failed! Error number(1-60) "click middle button" to retry, hold  "back button" to abort ". this happens on the mobile way on my IPhone(IOS). and seems like dragging the file under ext/apps/update takes long but when about to finish it says.", tried to uncompress and re compress. tried on mac iOS (source.zip) and on my app (.tgz), on the app either almost done or close update was not successful,i went to go look to see if firmware is complete; but seems like the proccess did not work as games has nothing in the game/ folder. second check to see if the firmware worked, and checked the GPIO/ folder to see if any esp's were in there nothing just says empty as GAMES/ does. not sure if the fact 0.71.1 update on the flipper I have. and the Rouge master update 0.71.2 has to do with anything. SoS.

### Reproduction

Mac iOS 
1. update 0.71.1 on flipper zero
2. download on Mac, source.zip 
3. unzip(decompress)
4. drag folder to SD card/update.
5. and this is where it takes a long to to upload and says" OPERATION ERROR(title) 
                 Current operation was interrupted connection to the device is lost.
                         -----------how to fix------
                              1. check usb connection
                              2. ensure that flipper is not locked with pin code.
                              3. connect your flipper in update and recovery mode and start repair.
            -----------------------------------------------".

mobile app iOS*
1. go to releases 
2. click on the rouge master 0.71.2
3. download and look for the file 
4. click share to flipper zero
5. and click UPLOAD. 
6. and it will continue updating and usually this where it will say "Update Failed! Error number(1-60) "click middle button" to retry, hold  "back button" to abort ".

### Target

loading

RogueMaster, 2022-11-20:
You're doing it wrong. Please read some ReadMe's or watch some How To's... extract TGZ and put into SD card/update

closed 2022-11-20 by RogueMaster

Yourit7870, 2022-11-20:
got it working thanks for the support! love what you do!