RogueMaster/flipperzero-firmware-wPlugins #438: Feature request disable custom animations

Darkcast, 2022-11-23:
### Description of the feature you're suggesting.

ive have noticed that some of the custom disable custom animations are not well optimized and some time it freezes my flipper. it would be nice if there were a way to disable custom animations and just leave stock one. 
thanks

closed 2022-11-23 by RogueMaster

Schecher1, 2022-11-23:
> You can do this by changing the dolphin folder on the SD card

Do you happen to have the stock folder with the standard images/animations for the dolphin ?

RogueMaster, 2022-11-23:
See stock firmware

RogueMaster, 2022-12-05:
Exactly why you can go back to stock firmware, I don't need you using RM version.

RogueMaster, 2022-12-05:
Just stay on stock, don't switch back

Schecher1, 2022-12-05:
why are u so mad? I just wanted the normal dolphin back and not this goku or whatever his name is

RogueMaster, 2022-12-05:
> why are u so mad? I just wanted the normal dolphin back and not this goku or whatever his name is

You didn't give the stupid response, the other guy did. To solve your problem:
- Install official firmware (as previously mentioned)
- Copy dolphin folder from SD card to computer
- Install RM firmware again
- Copy back/Replace the dolphin folder from computer to SD card

Schecher1, 2022-12-05:
I already did 12 days ago...

RogueMaster, 2022-12-05:
Ok good, the response was meant for the moron that doesn't see help when he gets it

Schecher1, 2022-12-05:
Ah now I saw his answer as well, I thought you meant me