RogueMaster/flipperzero-firmware-wPlugins #455: One of the Favorite Apps not working

mihail4anov, 2022-11-27:
### Describe the bug.

Hello.
I am on the RM11250310-0.71.2-0e622a2.
Tried to set up primary and secondary favorite app, but works only one of the buttons.

### Reproduction

I have set Sub-GHz as Primary Favorite app and iButton as Secondary favorite app.

So, Sub-GHz opens as it should (while I hold down button),
but iButton does not start while I hold up button.

Then I changed Secondary favorite app to Applications and it started to work. (RFID Fuzzer opens as it sould, while I hold up button).

RogueMaster, 2022-11-27:
hold up and hold down

closed 2022-11-27 by RogueMaster

mihail4anov, 2022-11-27:
> hold up and hold down

What is what I was doing. One of the buttons does not work as it should.

blacklabel2020, 2022-11-27:
I thought favorite app was left and second favorite app was hold left