RogueMaster/flipperzero-firmware-wPlugins #458: SubGhz issue with specific remote

kritonpc, 2022-11-29:
Although I've never had any issues with sub ghz and I have RogueMaster FW with the correct config, I encountered an issue today.
I recorded a remote that works at 433.92 and when i pressed send directly after recording it and it was recognized, it showed me a transmission error saying that the specific frequency is not supported (although it is 433.92). I saved it and tried sending it this wat and when selecting the file it shows the error "Cannot parse file". I tried it several times but it doesn't work. When saving it RAW of course it worked, but the rest seems odd. Any idea what can be wrong? Here is the .sub file as it was saved on flipper:

```
Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: CAME
Bit: 32
Key: 00 00 00 00 66 CF 63 02
```

RogueMaster, 2022-11-30:
Please test if this same problem happens on OFW

kritonpc, 2022-11-30:
I don't have a flipper with OFW. Can you test if the parse error happens in another flipper too?

RogueMaster, 2022-11-30:
If you don't care to try then don't post here

closed 2022-11-30 by RogueMaster

kritonpc, 2022-11-30:
It's not the easiest thing to switch firmwares without losing anything

RogueMaster, 2022-11-30:
I don't know what device you are using, but on the Flipper Zero, it is the easiest to switch and not lose anything

kritonpc, 2023-01-29:
I recently encountered the same issue with another remote. I get the same errors. I tried it on both RogueMaster FW and OFW and it has the same behavior. The new remote is:
```
Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433889000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: CAME
Bit: 32
Key: 00 00 00 00 F7 46 88 12

```

RogueMaster, 2023-01-29:
If you have an issue also happening with OFW please open an issue there.

kritonpc, 2023-01-29:
@RogueMaster Thanks for the reply. I opened it there too so... Let's see.