RogueMaster/flipperzero-firmware-wPlugins #460: Infrared Slow Launch

m5performance, 2022-11-30:
### Describe the enhancement you're suggesting.

I have noticed since Infrared has been built as a FAP and moved from being internal app to external app the launch became very slow. Suggestion to move it back as internal app to improve launch speed.

RogueMaster, 2022-11-30:
This is a known issue I can't fix for unless I make it internal again; not trying that for now

closed 2022-11-30 by RogueMaster

m5performance, 2022-11-30:
Noted, Thank you for your continuous support to keep this repo updated :) Would you be kind enough to guide me on moving it to internal or if there is any documentation that I could refer to?

RogueMaster, 2022-11-30:
undo https://github.com/RogueMaster/flipperzero-firmware-wPlugins/pull/398 and parts of https://github.com/RogueMaster/flipperzero-firmware-wPlugins/pull/405

m5performance, 2022-11-30:
Thank you so much