RogueMaster/flipperzero-firmware-wPlugins #463: Cannot parse file

aberg94, 2022-12-10:
### Describe the bug.

Verified file was accurate according to code listed here, I have also removed it from my SD and reapplied it. Its the only file I have this issue with. File is currently in the default path of /any/subghz/Misc/Crosswalk_Unknown.sub

I am running the latest RM firmware 0.72.1, installed after upgrading dev firmware 0.72.1.

### Reproduction

1. Turn on Flipper
2. Head to SubGhz folder
3. Attempt to run Crosswalk_Unknown.sub
4. Cannot parse file

### Target

/any/subghz/Misc/Crosswalk_Unknown.sub

ClaraCrazy, 2022-12-12:
`Bits: 29` is the issue. Should be 24. Works then

RogueMaster, 2022-12-12:
Thank you! @ClaraCrazy