RogueMaster/flipperzero-firmware-wPlugins #469: Help file reader standard proposal.

NF8C, 2022-12-14:
### Description of the feature you're suggesting.

I propose a standard system for reading help files and a way to include them.

1. Help files would be text files with the extension .nfo
2. The Flipper would not display the help files when browsing the menu.
3.  The Flipper would have a built in help file reader.
4. The reader would would be activated by highlighting the program or folder then pressing the left arrow.
5. When activated the help file viewer would attempt to open a file with the .nfo extension that matches the name of the file or folder that was highlighted.
6. If no matching .nfo file was found the reader would simply display "No help file found"

RogueMaster, 2022-12-14:
> 3\. The Flipper would have a built in help file reader.

this is working already :)

RogueMaster, 2022-12-14:
This is an interesting feature proposal, however, I don't really like the idea of another text type to add, nor do I like having text files that can't be read by flipper either. Also, not sure what the help contents would be.

closed 2022-12-14 by RogueMaster