RogueMaster/flipperzero-firmware-wPlugins #476: Settings - Desktop - Favorite Apps issue

bip3r, 2022-12-22:
### Describe the bug.

I there..
I found some issues with the latest Flipper firmware..

In the ReadMe.md of  flipperzero-firmware-wPlugins repository in the section "All Changes/Features" it says:

Settings: Updated HOLD UP to go to Primary Favorite [(By RogueMaster)]
Settings: Updated HOLD DOWN to go to Secondary Favorite [(By RogueMaster)]

but what I found is that within the menu:
Settings ->  Desktop -> Primary Favorite App
when you enter it, above it says "Secondary favorite app:" 

and the same in 
Settings ->  Desktop -> Secondary Favorite App
when you enter it, above it says "Primary favorite app:" 

When I try to set the Primary Favorite App (HOLD UP), it sets the Secondary (for HOLD DOWN)
And opposit with the other menu.

Moreover, after one change of the fav app, I'm not able to set any of the availible apps. Only Sub-GHz, NFC and 125kHz work, while with e.g. Infrared, HOLD UP/DOWN does nothing.

Anyway, that firmware is just fabulous! Thx ofr a great job.

### Reproduction

Enter Settings -> Desktop -> Primary/Secondary Favorite App

### Target

latest Flipper RM firmware

RogueMaster, 2022-12-22:
For any apps that are available in FAP form under Applications => Main, they should be set there instead of from main menu. Will check your findings to see if something can be fixed with the SWAP on the settings set.

bip3r, 2022-12-22:
Thx! Indeed this way (Applications => Main) it works.
So the only minor issue is the swap of Primary/Secondary Fav App menu.

RogueMaster, 2022-12-27:
The settings are fixed. The issue is still there kind of. But up is Primary and down is Secondary on latest release.

closed 2022-12-27 by RogueMaster