RogueMaster/flipperzero-firmware-wPlugins #477: Frequency analyzer

sky7d, 2022-12-25:
### Describe the bug.

In the last fw frequency happens that frequency analyzer doesn't work anymore...in the net many devices have this kinf of problems

### Reproduction

Happen suddendly and the frquency analyzer doe sn't mwork anymore

### Target

Frequency analyzer

RogueMaster, 2022-12-25:
> in the net many devices have this kinf of problems

This is not true. I have tested with another flipper and the Frequency analyzer works perfectly.

closed 2022-12-25 by RogueMaster

sky7d, 2022-12-25:
> <RogueMaster/flipperzero-firmware-wPlugins/issue/477/issue_event/8108629634
> @github.com>

I found several people in different forums who had the same problem as me
so i reported the bug. I reinstalled the original fw and the frequency
analyzer came back to life...

                            Merry Xmas Marco

RogueMaster, 2022-12-25:
Happy Holidays Marco! Other peoples issues won't necessarily be your own. If those people truly have issues, why aren't they in my Discord instead of some random forum. 

If you are on the latest firmware, the Frequency Analyzer is tested to be working. Both with another flipper and my car key that has a weak signal.