RogueMaster/flipperzero-firmware-wPlugins #479: NFC Writing

TTVDannyg82, 2022-12-28:
### Describe the bug.

When writing to an NFC (any type), it will write on writing device, but when you leave emulation, it reads "Can not save key file", I have tried this on unleashed and official firmware and i do not get this issue

### Reproduction

1. Click ok
2. Go to NFC
3. Click add manually
4. Click NTAG215 or NTAG216
5. Click more
6. Click save
7. Name it
8. Click on the nfc file
9. Click emulate
10. Use NFC Tools (ios or android) to write a text value (or any)
11. Click back

RogueMaster, 2022-12-28:
[RM-NFC-FIX.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10316800/RM-NFC-FIX.zip)

Could you try this version and see if it gives the same issue?

RogueMaster, 2022-12-29:
Latest source update should fix it, please let me know if the problem persists. Update Release For Current Changes Coming Very Soon. I reverted some draft PRs in order to keep NFC the same as unleashed. I will re-import them when confirmed successful by OFW or the problem is better identified.

- Reverted NFCV, NFCA, SLIX and MRTD to fix issue with NFC
- - Reverted: [Mifare dictionary attack performance improvements. #2173 (By nezza)](https://github.com/flipperdevices/flipperzero-firmware/pull/2173)
- - Reverted: [NFC - Machine Readable Travel Documents #1866 (By qistoph)](https://github.com/flipperdevices/flipperzero-firmware/pull/1866)
- - Reverted: [(WIP) added ISO15693 reading, saving and revealing from privacy mode (unlock) #1991 (By g3gg0)](https://github.com/flipperdevices/flipperzero-firmware/pull/1991)

closed 2022-12-29 by RogueMaster

RogueMaster, 2022-12-29:
This new build has it fixed without all the reverts, please test: 

[RM-NFC-UNREVERTEDandFIXED.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10322084/RM-NFC-UNREVERTEDandFIXED.zip)

BuggyDriver, 2023-08-01:
This problem appears again in the latest releases