RogueMaster/flipperzero-firmware-wPlugins #480: "22798 [DEV] Full Update @Qu3gs4r ERROR: Cannot find update directory 22798 [BKD] Current operation finished with error: "Cannot find update directory"

harrywillss, 2022-12-29:
Getting the error "22798 [DEV] Full Update @Qu3gs4r ERROR: Cannot find update directory
22798 [BKD] Current operation finished with error: "Cannot find update directory"" when trying to update through qFlipper. Unable to update on IOS too as it fails after 50%. All files seem to be where they are supposed to be on the flipper.

Whit3HatRabbit, 2022-12-29:
I had to remove update files from the SD card on the flipper. Reupdate to current stock firmware then add Rogue Master to your flipper zero update folder. Then on your device go to the update fold you just put rogue master in click the folder and you'll see an up arrow next to update click it and you should be good!🤙
Hope it helps!

RogueMaster, 2022-12-29:
> I had to remove update files from the SD card on the flipper. Reupdate to current stock firmware then add RogueMaster to your flipper zero update folder. Then on your device go to the update folder you just put RogueMaster in click the folder and you'll see an up arrow next to update click it and you should be good!🤙 Hope it helps!

 💯 Thank you! 😄 This should work!

- Extract/unzip the downloaded file from above to a folder on your computer (Use 7Zip or WinRAR)
- Copy the entire folder to the update folder onto the SD card you're using in your FlipperZero
- Once the SD card is back in the Flipper (or you've copied the files successfully via qFlipper), press down at the Desktop (the screen with the animations)
- This opens the Archive app, press left/right to get to the Browser section
- Locate the update folder, and the folder you copied earlier
- Locate the update file and select it
- Select Run in App

closed 2022-12-29 by RogueMaster