RogueMaster/flipperzero-firmware-wPlugins #494: How to verify NRF24 is correctly connected?

exander77, 2023-01-09:
### Description of the feature you're suggesting.

Apps seem to load fine and do something even if I do not connect anything. So I don't really know if it is working or not.

jilleb, 2023-01-09:
Try a vulnerable mouse + dongle close to it, and sniff to see if you can find anything. Note: you will not find much when the mouse isn't used.

RogueMaster, 2023-01-09:
If it changes to Sniffing=Yes, it is connected and working

closed 2023-01-09 by RogueMaster

exander77, 2023-01-09:
> If it changes to Sniffing=Yes, it is connected and working

Sadly no. If that was the case I would not be complaining. Without anything connected, sniffing can be changed to Yes.

RogueMaster, 2023-01-09:
I see, that used to not be the case but you are right in that it changes now. @mothball187 any availability to update the NRF Sniffer?

exander77, 2023-01-10:
> I see, that used to not be the case but you are right in that it changes now. @mothball187 any availability to update the NRF Sniffer?

If that should work than this is actually bug report and not feature request.

mothball187, 2023-01-10:
@exander77 @RogueMaster if you see possible addresses being displayed in the app when it is sniffing, then that means the radio is connected properly. You should see possible addresses in most environments, as the radio is sniffing data in the 2.4GHz range which is extremely popular. Bluetooth, Wifi, zigbee, etc. all operate in that range.