RogueMaster/flipperzero-firmware-wPlugins #497: Bluetooth BadUSB

yodaluca23, 2023-01-11:
### Describe the enhancement you're suggesting.

Let us run normal BadUSB scripts, over Bluetooth, it should definitely be possible, since we already have BadUSB and a Bluetooth controller keyboard. It just needs to be combined! BadBluetooth?

### Anything else?

Related Issue: https://github.com/flipperdevices/flipperzero-firmware/issues/2294

RogueMaster, 2023-01-11:
Cool, yeah I'm interested... see what you can find and PR if you get it figured out :)

I can't dedicate time to this right now, I just got sick and have a few other projects on my hands...

RogueMaster, 2023-01-12:
Keep in mind you can't force your flipper to pair to a PC so the hacking part of the badusb is kind of removed...

RogueMaster, 2023-01-19:
I'm closing this for now since I don't know how to work past the unpaired BT badusb running. Please do feel free to open a PR if you find a solution.

closed 2023-01-19 by RogueMaster