RogueMaster/flipperzero-firmware-wPlugins #506: Physically can't install it

GingyMaster, 2023-01-17:
### Describe the bug.

Every time that I try to install it on the last step (using the flipper's browser) I always get either error 9 99 or 1 60. I'm not sure if this is just an issue with my flipper or computer either.  

### Reproduction

1. Use the flipper's browser
2. Go to updates
3. Press the Rogue Master file
4. Use the update thing that you use to actually install the firmware.
5. Get either error 1 60 or 9 99.

RogueMaster, 2023-01-17:
Hi, please install OFW and then install RM. I've seen this happen when the previous firmware was too large. You may even need to DFU install RM to the run the installer to get assets.

closed 2023-01-17 by RogueMaster

GingyMaster, 2023-01-18:
Hi, I'm sorry to bother you again and I know it sounds stupid, but I don't
know what OFW is. I'm quite new to the flipper and I don't know anything
about it. Sorry, but do you mind giving me a link or something to install
it? Sorry about my inexperience.

RogueMaster, 2023-01-18:
OFW is the original firmware by flipper devices

GingyMaster, 2023-01-18:
That's weird, I already have the original firmware installed but it still
won't work. Sorry for being such an annoyance but is there anything else I
can do?

RogueMaster, 2023-01-18:
Install Original Firmware again via qFlipper, then follow install instructions for RM here: https://github.com/RogueMaster/flipperzero-firmware-wPlugins#install .

GingyMaster, 2023-01-18:
I just tried that and I'm still getting error #[1-60].

RogueMaster, 2023-01-18:
You may have an error with your SD card, not really sure what that error stands for. Try installing DFU via qFlipper and moving files from resources.tar manually onto your SD card

GingyMaster, 2023-01-23:
Hello again, I thought that you were probably right about it being an issue
with the SD card so i ordered a new one and it worked! Thank you very much
for the help.