RogueMaster/flipperzero-firmware-wPlugins #512: NFC-V emulating error

danidemis, 2023-01-26:
### Describe the bug.

Hi, I am experiencing a problem with the emulation of an NFC-V. In RM version 0.73.2 [13 dec 2022], build 20a0a43d2 [5222] 1.13.3:L [7] dev I had managed to scan an NFC-V, save it and emulate it functionally (so the original tag reader recognized it as the original badge). From a few versions after that I can do the tag reading, but at the time of emulation it is not detected by the reader. 

Does anyone have any way of knowing why it doesn't work as before? I tried reloading the old FW and it works, so I thought it was a problem related to updating the (.fap) NFC application. 

Thanks you all.

### Reproduction

1. Read a NFC-V tag (in my case a ISO15693 SLIX-L)
2. Try to emulate in front of the same reader that recognize the original NFC-V tag
3. The reader do not trigger in front of FZ emulation

g3gg0, 2023-01-26:
can you give some more details?
 - does the SLIX-L have a password set?
 - does the reader use passwords?
 - can you share the nfc file?
 - do the "old" nfc file and the "current" nfc file differ or are they the same?
 - 
regards,
g3gg0

danidemis, 2023-01-27:
Hi, thank you for your response. 
So to answer in order:

- No, no password was set, I think. In the old release where it worked I only had to read the tag with the FZ, I didn't have to read the reader or set a password for it to work.
- I don't know if the reader has a password.
- Yes, I am attaching the contents of the two files (one scanned with the old release and one with the new one). The content seems to be identical in the two files anyway (I mask only the UID, but I assure you they are the same)

[old_NFC-V.nfc.txt](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10517720/old_NFC-V.nfc.txt)
[new_NFC-V.nfc.txt](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10517721/new_NFC-V.nfc.txt)

Thanks, best regards.

danidemis

g3gg0, 2023-01-27:
Thanks,

if you emulate the card, does the Log button appear and show you some of the commands?

regards,
g3gg0

g3gg0, 2023-01-27:
Also, I have made a few changes to the emulation code lately.
They seem to be stable enough to try, maybe you want to give them a try, too.
Perhaps this fixes the issue for you as well.

This is already integrated into latest RM420 
[f7-update-RM420FAP.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10518857/f7-update-RM420FAP.zip)

RogueMaster, 2023-01-28:
If emulation changes were made recently and are already in RM, it is likely these changes @g3gg0 are what caused this issue, no?

RogueMaster, 2023-01-28:
@danidemis you have an install for build 20a0a43d2 that you can still test with and confirm it works with that?

g3gg0, 2023-01-29:
> If emulation changes were made recently and are already in RM, it is likely these changes @g3gg0 are what caused this issue, no?

well, there were various additions and changes. we do not know what is wrong.
it works here fine, when using a consumer device with integrated SLIX-L reader and also with a proxmark 3.
however there was a missing feature when reading multiple blocks, where the card didnt respond with the lock bits the reader expected.
added this recently, along with a few other optional commands, SLIX implements.

So the best is, we try with the latest version. here we should dig deeper into what really happens.
(i.e. what does the reader expect, what does the flipper reply etc)

g3gg0, 2023-01-29:
and for clarification: tried the claimed revisions of course and all of them work as expected with my 2 readers.
so we have to find out what the reader is asking for, the emulator cannot serve.

danidemis, 2023-01-30:
> Also, I have made a few changes to the emulation code lately. They seem to be stable enough to try, maybe you want to give them a try, too. Perhaps this fixes the issue for you as well.
> 
> This is already integrated into latest RM420 on a private repository and will get PR'd/merged soon. [f7-update-RM420FAP.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10518857/f7-update-RM420FAP.zip)

I tried flashing this firmware but the problem persists.

danidemis, 2023-01-30:
> @danidemis you have an install for build [20a0a43](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/commit/20a0a43d2502b071ba0ee61509104d86ac6e3d5a) that you can still test with and confirm it works with that?

Yes, I removed the apps of newer version, flashed the RM version 0.73.2 [13 dec 2022] build 20a0a43 [5222] and again the FZ resumed emulating the file correctly.

g3gg0, 2023-01-30:
### sniffing reader + original-card
okay thank you. can you please
a) enable Debug in Settings -> System
b) delete the \nfc\debug.txt 
c) run NFC -> Extra Actions -> Listen NfcV Reader
d) hold the card on flipper's back
e) put card+flipper on the reader

you should now see log messages on the screen and also a populated debug.txt, which i need.

### emulating card
also when emulating, the screen should a log (press center button)
can you
a) show me a video what it shows when you emulate your card
b) also delete, emulate and send me the debug.txt

thanks!

danidemis, 2023-01-30:
Hi, I followed the instructions you gave me.  Here are the two debug files (one tag+FZ in front of the reader, the other while I'm emulating the file).

I'll give you the video tomorrow, today I forgot to do it 

Thanks
[reading-tag+FZ-debug.txt](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10538262/reading-tag%2BFZ-debug.txt)
[emulating-debug.txt](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10538264/emulating-debug.txt)

g3gg0, 2023-01-31:
okay i identified an issue that could result in some readers not to accept the card's response.
this looks like an issue with the modulation generation.
causes no severe problem on proxmark though.

why this reader however does an IDENTIFY, SELECT and READBLOCK, i do not understand right now.
i would have expected the reader to bail out in IDENTIFY already if this would be the root cause...

but let me fix this issue and we will do another test.

oh and you can delete the logs also, just in case.

danidemis, 2023-01-31:
Hi,
I did a test with the various RM releases. The latest version on which emulation works for me is 0.74.3 [26-12-2022] 499dea07f [5552] 1.13.3:L [7] dev . I will now try the latest available version that came out yesterday and update this comment. 

Thanks, best regards.

g3gg0, 2023-01-31:
there was nothing changed in that regard, so i can anticipate the results :)

danidemis, 2023-01-31:
For the time being I will keep the old release that is convenient for me to be able to emulate that tag.

I really thank you for your support 🙏

Best regards.

g3gg0, 2023-02-01:
I fixed two issues:
some edges were "flipped" when sending sequences. this could cause reading errors.
however the proxmark3 had no problem with it. still this is not as it should have been.
fixed that. that took quite long, i hope this was the root cause.

another issue was the "frame delay time", i.e. the time the tag waits before sending its response.
the original tags were a bit slower than my code. so i increased the wait delay a bit so it matches the original tags.
e.g. toniebox is very sensitive to that timing, but it worked well with how it was.

[f7-update-RM420FAP.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10552149/f7-update-RM420FAP.zip)

can you try with that version, if it helps?

danidemis, 2023-02-01:
Hi,
I will try as soon as I have a chance, which is Friday, February 3. In the meantime, I thank you for your support.

Best regards

danidemis, 2023-02-03:
Hi, 

I installed the version you provided but unfortunately the FZ in emulating mode is still not "seen" by the reader. 
To be scrupulous, I also recreated the file to be emulated by reading the original tag again.

If you have other actions for me to take let me know.

Best regard

g3gg0, 2023-02-03:
this is not what I expected... :(
can you again sniff putting the original card on the reader *multiple times*
in the last log i have only seen a INVENTORY (01) and then a SELECT (25) by UID
[image]

maybe you can play a bit with the card position when moving into field.
or put the FZ onto the reader's side and then move the card into the field.

in the log you should see not just INVENTORY or SELECT, but also READ BLOCK

g3gg0, 2023-02-03:
the stupid thing is, i cannot make out any difference between the old version and the new when testing 
all possible commands with a proxmark 3 or the toniebox. 
also the picopass emulation from @nvx didn't seem to have noticeable problems, so hmm, idk.

g3gg0, 2023-02-03:
oh and please provide a debug.txt with the OLD version too.
the old revision, where emulation works, should also support debug.txt logs.

danidemis, 2023-02-03:
I'll share these two debug files with you, one with the version running as I do the emulation (the debug-working.txt) and the other with the latest version you provided me with the sniffing mode (debug.txt). In performing the sniffing, I approached the reader 3/4 times by snapping the lock with the original tag.

I hope they can help in some way :)
[debug_working.txt](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10579301/debug_working.txt)
[debug.txt](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10579304/debug.txt)

danidemis, 2023-02-03:
Another one file with FZ near the reader when unlocking door with original tag:
[debug-FZ-near.txt](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10579368/debug-FZ-near.txt)

nvx, 2023-02-03:
I'd noticed some weirdness with picopass too but hadn't confirmed it wasn't my problem before harassing g3gg0. Have passed along the results of my testing which will hopefully yield some results. Bit of an odd one.

g3gg0, 2023-02-03:
it looks like the stopping/restarting DMA after every signal to form a sequence is the issue here.
i introduced sequences, because with NfcV every bit has 16 polarity changes, so every bit consumed 64 byte.
this are 512 byte of memory per byte response length. maximum response length is 1024 or so...
you see this doesn't scale well.

so i introduced sequences, where you at least have 8 bits per bit compared to 64 before. at least a bit better now.
however i have to restart the DMA after every transfer, why i had to use assembly code to keep up the speed.
this was subject to change before getting merged, so i optimized it and could get away with a small C routine.

however it seems that C routine isn't fast enough in every situation *and* there is a cummulative error, causing longer responses to not get accepted.

okay. guess it will take another 12 hours of coding/testing to change that to.. hmm maybe a ringbuffer DMA?

g3gg0, 2023-02-04:
okay, what about this variant?

[f7-update-RM420FAP.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10593549/f7-update-RM420FAP.zip)

danidemis, 2023-02-06:
> okay, what about this variant?
> 
> [f7-update-RM420FAP.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10593549/f7-update-RM420FAP.zip)

hello, I also tried to install this latest version you gave me. The emulation still doesn't work, I attach you the emulation debug. I also made a new scan&save of the original tag, I noticed that there is an extra line:

[image]

I will provide the sniffing log later if you think it would be helpful.

[debug-06022023-v2.txt](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10663580/debug-06022023-v2.txt)

g3gg0, 2023-02-06:
that pm3 easy is more than good enough.

# first get your card's response timings
can you run a 
```hf 15 cmd read u 0``` (or the appripriate command for that task in your version)
and then
```hf list 15 -f``` (to show the FDT and decode the instructions as ISO15693)
you then can do the same for the flipper emulating your card, and check if you spot any difference in FDT od answer?

danidemis, 2023-02-06:
OK, I'll catch up on this information ASAP. 

Thanks

nvx, 2023-02-07:
> `hf 15 cmd read u 0` (or the appripriate command for that task in your version)

Assuming that's read unaddressed block 0, for iceman firmware the command would be `hf 15 rdbl -b 0 --ua`

If not unaddressed, drop the --ua flag.

RogueMaster, 2023-02-16:
Any updates for this issue? Should I close it?

danidemis, 2023-02-16:
I have not yet managed to read with the proxmark 3 easy. I will post the reading asap.

danidemis, 2023-02-22:
Hi, 

sorry for your waiting. I provide to you these files. I'm not skilled with proxmark, so I don't know if these can help you...

-  **card_output** contain proxmark command for the original tag info
-  **flipperEmulating-oldVersion-WORKING_output** contain proxmark command for the FZ with working firmware 
-  **flipperEmulating-newVersion-NOT_WORKING_output** contain proxmark command for the FZ with not-working firmware

Perhaps tomorrow I will be able to provide you with the sniffing of the communication between FZ and reader and between original tag and reader.

Thanks

g3gg0, 2023-03-19:
any news on the sniffing?

RogueMaster, 2023-03-29:
@danidemis When you can, please continue this discussion with @g3gg0 on our Discord...

[image]

closed 2023-03-29 by RogueMaster

g3gg0, 2023-05-10:
although this is vanilla flipper zero with ISO15693 patches,
can you test if this version is any better?
 [f7-update-local.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/11446606/f7-update-local.zip)

g3gg0, 2023-05-10:
@danidemis

g3gg0, 2023-05-10:
if this didn't do, maybe try this one:
[f7-update-local.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/11446767/f7-update-local.zip)

danidemis, 2023-05-11:
Hi, I have tried both firmwares but there is no improvement. In the next few days I will try sniffing the tag reading, unfortunately I had to leave the thread a bit because I haven't had a chance until now to use the badge.

g3gg0, 2023-05-11:
i guess setting up an RF probe and a scope might not work because this is the main door of your company or such...? ;)
alternatively play with the plot function of proxmark, if there is a chance to see the difference in reader<->card communication.
didnt try this.

as you can see here: https://github.com/flipperdevices/flipperzero-firmware/pull/2316
the SLIX-L emulation on RF field level looks very good and matches an original SLIX-L well.

on protocol level we have already seen that there is no difference. 

so either assumption might be wrong.

MadjidZ, 2023-05-11:
I noticed that for some NFC, I managed to copy the card but in emulation via the FZ the reader did not detect it.

On the other hand, by copying via NFC Magic from my FZ to a card as can be found here (https://lab401.com/fr/products/mifare-compatible-1k-uid-modifiable-pack-of-5), the reader was good at reading it.

Looks like as you assume that in emulation the FZ suffers from noise...

PS : I just installed [this version](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/11446767/f7-update-local.zip) I'll get back to you tomorrow

g3gg0, 2023-05-11:
this is about ISO15693 emulation, not mifare classic.

MadjidZ, 2023-05-12:
It's in the title of the page and I read the content well, of course I know we're talking about NVC-V here... Read again what I'm describing, I'm simply explaining that the probleme " noise" on the emulation seems to reproduce on other NFC specifications via the FZ...

By copying the NFC from the FZ to a physical card, the reader manages to read (and not from the FZ in emulation mode).

Anyway, I thought you might be interested...

g3gg0, 2023-05-12:
ok i just wondered why you thought mifare is a thing.
well that noise isn't that intense that i'd expect issues with it as its quite low frequency.

g3gg0, 2023-05-12:
[f7-update-local.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/11465912/f7-update-local.zip)
@danidemis any better with this?

danidemis, 2023-06-14:
Hi,

installing the latest update (build RM0607-1145-0.84.3-81c9df5) of Rogue Master the problem seem be fixed!! I had to rescan the NFC tag another time but now i can emulate without any problem.

Now you mark this as definitely closed.
Thanks for your support over the time.

danidemis

g3gg0, 2023-06-14:
I guess it was the same delay that reduced readability for android phones which @nvx fixed recently

nvx, 2023-06-14:
> I guess it was the same delay that reduced readability for android phones which @nvx fixed recently

I love it when a fix for one thing ends up fixing other stuff too 😁