RogueMaster/flipperzero-firmware-wPlugins #516: BadUsb crases

TheTomgamerboi, 2023-01-28:
### Describe the bug.

Flipper crashes when attempting to run several badusb scripts after each other

### Reproduction

1.Execute badusb script
2.Go back to badusb menu
3.Open another badusb script
4.Crash?
5.retry  2-3 if no crash with diffrent scripts

RogueMaster, 2023-01-28:
Can you check if you can re-create on OFW? How many does it take to crash? How many to re-create on RM?

TheTomgamerboi, 2023-01-30:
1. Does not occur on OFW
2. Crashes upon opening the second script, sometimes more. Execute first script, close that script, open new script, instant crash. All manor of crash reasons.

I'm also running my own fork of the firmware with the only change being the default animations.
Annnd now that I've re-rerun the update file to go to my RM CFW the issue has disappeared. fml

closed 2023-01-30 by TheTomgamerboi