RogueMaster/flipperzero-firmware-wPlugins #520: Device freezes after opening dolphin status screen

Elletre, 2023-02-03:
### Describe the bug.

On the few last builds I've faced the problem that the whole flipper is freezes after pressing the right arrow from the main screen. Only a hard reboot with the back+left arrow helps.

### Reproduction

1. Open the main screen
2. Click the right arrow on the joystick

Can be reproduced around one time of the three

RogueMaster, 2023-02-05:
A solution is being worked on for this :)

ESurge, 2023-02-09:
This should be fixed in [#527 ](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/pull/527)

Probably be released within a day

closed 2023-02-09 by RogueMaster