RogueMaster/flipperzero-firmware-wPlugins #526: Sub-ghz Outside range allowed...

Gadgetroch, 2023-02-09:
### Describe the bug.

I receive 300 mhz sub-ghz, and i can send it, which is not allowed
I have no country code in "about" menu, i just get "--"

It's dangerous and need to be locked by country

### Reproduction

1. Get not allowed mhz signal
2. Send it

Gadgetroch, 2023-02-09:
> Please open the extend range app and send a screenshot of the settings

Hi,

All is false...

Gadgetroch, 2023-02-09:
HW Version Info :
12.F7B9C6 R01:-- Amipode

Risky Range : No
Ignore Region : No

RogueMaster, 2023-02-09:
What region are you in?

Gadgetroch, 2023-02-09:
France

maede97, 2023-02-10:
Can reproduce (both settings at "no", with region being -- as well, should be EU).
https://docs.flipperzero.one/sub-ghz/frequencies shows that it should actually show a screen that this frequency is not allowed.
Unleashed Fork has this in the docs: https://github.com/DarkFlippers/unleashed-firmware/blob/dev/documentation/DangerousSettings.md

RogueMaster, 2023-02-11:
Could one of you please try these to see if it fixes the issue?

https://lab.flipper.net/?url=https%3A%2F%2Frogue-master.net%2F%3Ffile%3DRM0211-0.76.1-RegionFix.tgz&channel=RM0211-0.76.1-RegionFix&version=0.76.1

https://rogue-master.net/?file=RM0211-0.76.1-RegionFix.tgz

Gadgetroch, 2023-02-11:
I m in holidays... I can test it Monday

RogueMaster, 2023-02-11:
I was able to test with some TPMS files... prior to the fix, on OFW, and after this fix. I think this is fixed. Please let me know if you have a different experience.

closed 2023-02-11 by RogueMaster