RogueMaster/flipperzero-firmware-wPlugins #529: After update 303.875 shows as Transmission is blocked

vaidasnet, 2023-02-13:
### Describe the bug.

303.875 MHz which used for customer assistance buttons (Lowes, Walgreens) doesn't work anymore, shows: Transmission Blocked
It worked in previous version, I didn't make any changes to **setting_user.txt** or **extend_range.txt**

### Reproduction

1. open Sub-GHz
2. Saved
3. Choose "Customer assistance buttons"
4. go to "Lowes" or "Walgreens"
5. run any .sub here

vaidasnet, 2023-02-13:
NOTE: with unleashed firmware 303.875 MHz works fine

RogueMaster, 2023-02-13:
Use the Extend Range app to ignore the default region restrictions to allow this frequency

closed 2023-02-13 by RogueMaster