RogueMaster/flipperzero-firmware-wPlugins #530: updater_package unable to compile

alvarotorijano, 2023-02-14:
### Describe the bug.

This repo is not able to compile with the updater_package option. The oficial firmware does so it seems to be a problem with any change made in this repo. Or I can be missing something of course.
The error reported goes as follows: 

applications\settings\passport_app\dolphin/helpers/dolphin_deed.h:80:9: error: c
onflicting types for 'dolphin_deed_get_weight'
   80 | uint8_t dolphin_deed_get_weight(DolphinDeed deed);
      |         ^~~~~~~~~~~~~~~~~~~~~~~
In file included from applications\services/dolphin/dolphin.h:3,
                 from applications\settings\passport_app\passport.h:11,
                 from applications\settings\passport_app\passport.c:1:
applications\services/dolphin/helpers/dolphin_deed.h:80:9: note: previous declar
ation of 'dolphin_deed_get_weight' was here
   80 | uint8_t dolphin_deed_get_weight(DolphinDeed deed);
      |         ^~~~~~~~~~~~~~~~~~~~~~~
        CC      applications\plugins\nrfsniff\nrfsniff.c

### Reproduction

git pull --recurse / git clone --recurse this repo
./fbt.cmd (from windows) updater_package

### Target

updater_package

RogueMaster, 2023-02-14:
I can compile with no issues. Are you compiling on a fresh clone?

closed 2023-02-14 by RogueMaster

alvarotorijano, 2023-02-14:
I am with no success.

maede97, 2023-02-14:
Had the same issue. Fully DELETING the folder and cloning again (then building again) resolved the issue for me. Only deleting build and dist did not help!

alvarotorijano, 2023-02-26:
Please reopen this case. It is not possible to compile on windows using a fresh clone and running the line above