RogueMaster/flipperzero-firmware-wPlugins #537: NFC sector keys missing

drake4m, 2023-02-23:
### Describe the bug.

Hello,

The 2 latest builds have issues with the NFC emulating. I read several mifare classic 1k cloned cards and fobs, succeed to find all keys and read all sectors but when emulating keys are missing randomly in sectors and some are tagged as completely bad or missing. I COMPARED THE dumps with MCT.

### Reproduction

Read any mifare classic 1k
Emulate it (missing keys and sectors randomly)
Reader does not recognize the emulated card

### Target

NFC

RogueMaster, 2023-02-23:
RM has added NFC V, MRTD AND SLIX.... with that said... our Mifare Classic 1k code is the same as Unleashed... it is very odd it is not working... checking into this... will see if I can make a NFC fap to save some features to side load and revert to original code...

Wisiewig, 2023-03-06:
the reason for this is because of the reader and the copper coil on in the card, The flipper just isnt powerful enough to account for the differences in the frequencies, nor is it powerful enough to auto pwn a mifare classic 1k card

closed 2023-03-29 by RogueMaster