RogueMaster/flipperzero-firmware-wPlugins #539: Web Installer issue

Xbxjdjdjd, 2023-03-02:
### Describe the bug.

When trying to use web install, error "Failed to fetch firmware: TypeError: Cannot read properties of undefined (reading 'buffer')" appears (view attached photo)
[image]

### Reproduction

1. Go to web installer
2. Click install after connecting flipper via usb cable
3. error wil pop up

RogueMaster, 2023-03-02:
This is not an issue with the latest release. What file name are you trying to install?

closed 2023-03-02 by RogueMaster

JonathanERC, 2023-03-02:
Hello @Xbxjdjdjd I had the exact same issue, just download the ".tgz" package from the repository or Patreon page and then click "Install from File" in the Web Installer. It should install the update successfully. Hope it helps.

Thanks RogueMaster and Contributors for the new release.

RogueMaster, 2023-03-02:
If you can't post the FULL name of the file you are trying to install, please don't post.

JonathanERC, 2023-03-02:
> 

Hello, I just tried using latest GitHub release "RM0228-1236-0.78.2-5eb5e99" and Web installer in Google Chrome Version "110.0.5481.178 (Official Build) (64-bit)" with no extension and no cache, the same situation remains:
[image]
[image]

JonathanERC, 2023-03-02:
It worked like a charm now with "RM0228-1236-0.78.2-5eb5e99". Thank you so much.
[image]
[image]

Just to add something to the loop, the error is still present if you try to update to "RM0228-123**5**-0.78.2-5eb5e99" (That update file is where the GitHub hyperlink takes you to when you press "WEB INSTALLER" in the release of "RM0228-123**6**-0.78.2-5eb5e99"):
[image]

RogueMaster, 2023-03-02:
Oops, I see, fixed so both will work

MoScammin, 2024-03-31:
nope still same issue