RogueMaster/flipperzero-firmware-wPlugins #550: Cant emulate some subghz keys

Przemekmac1, 2023-03-08:
### Describe the bug.

Hi. I just downloaded github todays release and some keys report Error in protocol parameters description.  Ranges are unlocked, on previous version is was working. Its for eg. Harpoon keelow 64. 

### Reproduction

Go subghz, saved, select file, emulate

### Target

Fix bug

### Logs

```Text
None
```

### Anything else?

[image]
[image]
bplist00%C2%A3%01%02%03_%10%C3%87#path=any/subghz/Vitara.sub&Filetype=Flipper+SubGhz+Key+File&Version=1&Frequency=433920000&Preset=FuriHalSubGhzPresetOok650Async&Protocol=KeeLoq&Bit=64&Key=03+88+0C+AC+AB+EC+22+52&Manufacture=Harpoon_%10%12https://flpr.app/s%C3%90%08%0C%C3%96%C3%AB%00%00%00%00%00%00%01%01%00%00%00%00%00%00%00%04%00%00%00%00%00%00%00%00%00%00%00%00%00%00%00%C3%AC

RogueMaster, 2023-03-08:
Could you include the file?

Przemekmac1, 2023-03-08:
> Could you include the file?

Dump question, how to attach file to this post?

Przemekmac1, 2023-03-08:
[Vitara.sub.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/10920229/Vitara.sub.zip)
Ok, disregard stupid question . Here is file of my remote. I tried also to read orignal remote again and issue is the same.

RogueMaster, 2023-03-08:
Can you check if OFW has an issue with the file? If it is a file that requires a range not available, try with unleashed. I suspect this issue is present on other firmwares too.

Przemekmac1, 2023-03-08:
I think file is good. This is anyway on more different keeloq files. I downgraded to RM0228-1236 an its good. On RM0307-2236 its not. Unleashed (current) also works.

RogueMaster, 2023-03-08:
I made some [changes](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/commit/ca369f637d39b26a1cd2c0106aefab8baf9387f3), please see if this version fixes the issue... https://lab.flipper.net/?url=https%3A%2F%2Frogue-master.net%2F%3Ffile%3DRM0308-1303-0.78.2-ca369f6.tgz&channel=RM0308-1303-0.78.2-ca369f6&version=0.78.2

Przemekmac1, 2023-03-08:
Now it works (at lest no crashing, can't test now with receiver). 
Great job, thank you :)

closed 2023-03-08 by RogueMaster