RogueMaster/flipperzero-firmware-wPlugins #551: Dont delete custom animation manifest file

alvarotorijano, 2023-03-09:
### Description of the feature you're suggesting.

As user I don't want to be forced to rebuild my manifest for the desktop animations every time I install a new release from this repo. Is it possible to avoid this file to be overwritten during the installation?

RogueMaster, 2023-03-09:
This is a behavior adopted from the original firmware, it is not considered an issue. However, I may do something for next release to help with this...

RogueMaster, 2023-03-09:
Ok, next release will have something for you... thank you for contributing for the community!

https://github.com/RogueMaster/flipperzero-firmware-wPlugins/commit/cf4907a867c73830f299ec7328ff7b7c1c90204b

closed 2023-03-09 by RogueMaster

RogueMaster, 2023-03-09:
Looks like the new no animation build has a default to internal and doesn't use the manifest on the SD card for some reason... would be changes that were added by OFW not allowing it... not sure how to fix but open to PR requests