RogueMaster/flipperzero-firmware-wPlugins #586: Matrix animation displaying on flipper even when not in dolphin folder or manifest

Technically-Possible, 2023-04-01:
### Describe the bug.

When attempting to use the center button to swap animations on the flipper, the matrix animation is displaying even though it is not located in the dolphin folder or manifest. This is causing the flipper to freeze and crash. It seems that there may be a bug causing the incorrect animation to be displayed.

### Reproduction

Attempt to swap animations on the flipper using the center button.
Notice that the matrix animation is displayed even if it is not located in the dolphin folder or manifest.
Observe the flipper freezing and crashing.
Expected behavior:
The correct animation should be displayed when using the center button to swap animations on the flipper. The matrix animation should not be displayed if it is not located in the dolphin folder or manifest. The flipper should not freeze or crash.

Actual behavior:
The matrix animation is displayed even if it is not located in the dolphin folder or manifest, causing the flipper to freeze and crash.

### Anything else?

Additional information:

The flipper is running your latest version of the software.
The matrix animation is not located in the dolphin folder or manifest.
The issue is consistently reproducible.

qqmajikpp, 2023-04-01:
> ### Describe the bug.
> When attempting to use the center button to swap animations on the flipper, the matrix animation is displaying even though it is not located in the dolphin folder or manifest. [...]

The matrix animation is an internal animation and the only one. When this only plays, it is because the manifest is broken.

RogueMaster, 2023-04-01:
The no animations build has majiks blank animation instead of digital rain

closed 2023-04-01 by RogueMaster

Technically-Possible, 2023-04-02:
Im sorry, but maybe i didn't provide enough info, the other animation does play super rarely, here's a copy of my manifest if you can see something wrong please let me know.
[manifest.txt](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/11132478/manifest.txt)

qqmajikpp, 2023-04-02:
How about using one of the animation packs and using the included manifest in his awesome repo

Technically-Possible, 2023-04-02:
it was missing a few dolphin animations and I'm not a fan of the non dolphin animations

qqmajikpp, 2023-04-02:
https://github.com/RogueMaster/awesome-flipperzero-withModules/tree/rogue_main/graphics/dolphin-Stock