RogueMaster/flipperzero-firmware-wPlugins #590: No ESP32 apps on the wifi dev board using RogueMaster's firmware

Sh0ckFR, 2023-04-07:
### Describe the bug.

Hello,

I don't have the esp32 apps visible when I tried to install the roguemaster's firmware and I don't know why.

### Reproduction

1. Flashed the FlipperZero with RogueMaster lastest version
2. Flashed the ESP32 Wifi Dev Board with https://github.com/UberGuidoZ/Flipper/tree/main/Wifi_DevBoard/FZ_Marauder_Flasher and also tried the lastest release of https://github.com/FroggMaster/ESP32-Wi-Fi-Penetration-Tool/releases/

### Target

Flipper Zero

### Anything else?

When I check on my sd card in the GPIO apps I don't have the ESP32 apps.

RogueMaster, 2023-04-07:
reinstall using web installer

closed 2023-04-07 by RogueMaster

Sh0ckFR, 2023-04-07:
I will try it via the web installer, because I flashed it like 10 times already with qFlipper but nothing visible, I will let you know thank you

RogueMaster, 2023-04-07:
> I will try it via the web installer, because I flashed it like 10 times already with qFlipper but nothing visible, I will let you know thank you

This is why you don't see the proper apps... directions on readme don't include use of qFlipper for flash...

Sh0ckFR, 2023-04-07:
> > I will try it via the web installer, because I flashed it like 10 times already with qFlipper but nothing visible, I will let you know thank you
> 
> This is why you don't see the proper apps... directions on readme don't include use of qFlipper for flash...

Yep I saw that, but I use a new sd card bought on amazon like few days ago and I have the famous 0-99 error, the only decent way to flash the firmware in my case is to use the DFU mode to flash the original firmware, upgrading on your firmware, if I flash the original firmware and try to flash it on the web installer, I only have this 0-99 error, I think there is some issues with big sd cards on the flipper zero so I just bought a 16gb sd card to test it.

I will try to use the web installer when I receive the new card this sunday, but it could be interesting to have a copy of the esp32 apps somewhere if it possible of course.

By the way, thank you anyway for your great project!

RogueMaster, 2023-04-07:
you can also extract resources.tar manually to SD card

Sh0ckFR, 2023-04-07:
Awesome, thank you, I just saw this file, my bad, I not extracted the files (probably tired of the week haha) thank you mate, problem resolved :)