RogueMaster/flipperzero-firmware-wPlugins #595: Sub-GHz Remote Crashes on any .sub with custom modulation

Geofferey, 2023-04-16:
### Describe the bug.

Simple as title states. Whenever I attempt to play a .sub with:

    Filetype: Flipper SubGhz RAW File
    Version: 1
    Frequency: 313850000
    Preset: FuriHalSubGhzPresetCustom
    Custom_preset_module: CC1101
    Custom_preset_data: 02 0D 0B 06 08 32 07 04 14 00 13 02 12 04 11 36 10 69 15 32 18 18 19 16 1D 91 1C 00 1B 07 20 FB 22 10 21 56 00 00 C0 00 00 00 00 00 00 00

The issue pretty much affects any custom modulation or at least that's what I've discerned from difference in files that works vs. not. 

### Reproduction

1. Record a raw sub using a custom defined modulation scheme such as Hnd1
2. Setup a remote and point to recorded sub
3. Play the sub by pressing defined button
4. Transmission light blinks briefly, device crashes immediately and states:

   Flipper crashed and was rebooted
   NULL pointer dereference

### Target

bee09a44

### Anything else?

This is really a bummer because it basically cuts off ability to make Sub-GHz remotes for devices that flipper doesn't have inbuilt modulation for. If there is a better place to report issue or need anything else, please let me know.

RogueMaster, 2023-04-16:
Let's see if this is still a bug on the latest releases once you can access...

Geofferey, 2023-04-16:
@RogueMaster I just built and updated 0dacc776 and the issue is still present.

RogueMaster, 2023-04-16:
That .sub file you are using seems to be missing all the RAW_Data entries; does yours have them?

RogueMaster, 2023-04-16:
Honda Lock and Unlock buttons work on latest build, so far, I cannot replicate your issue but that file above is missing data...

Geofferey, 2023-04-16:
The data has been intentionally left out. I can tell you with 99% certainty the raw data is not problem... It's this:

    Preset: FuriHalSubGhzPresetCustom
    Custom_preset_module: CC1101
    Custom_preset_data: 02 0D 0B 06 08 32 07 04 14 00 13 02 12 04 11 36 10 69 15 32 18 18 19 16 1D 91 1C 00 1B 07 20 FB 22 10 21 56 00 00 C0 00 00 00 00 00 00 00'

More specifically this `Preset: FuriHalSubGhzPresetCustom`

Any file containing that setting will not work in the Sub-GHz remote app from the extensive testing I've done. I've gone as far as factory reset, sd wipe, restore OG firmware and re-flash of RogoueMastwer. 

Files containing `FuriHalSubGhzPreset2FSKDev476Async` or `FuriHalSubGhzPresetOok270Async` and similar work just fine. This is what I mean by "built-in" modulation(s)

To be 100% clear recording and playing . subs using the custom modulation works just fine, they only fail when being used with the Sub-GHz Remote. The issue is present in both your releases and unleashed.

RogueMaster, 2023-04-16:
Oh, so this issue is present only in the SubGHZ Remote, I see. I had mis-read that part... Nano from Unleashed is really good at working with Sub-GHz stuff. Perhaps he can try to take a stab at it. I know I can't do anything towards helping you with this, sorry

Geofferey, 2023-04-16:
I kinda had a feeling I wasn't in right place, haha. Can you point me to the original dev of this feature?  Or maybe even alert them to this so I am not duplicating all over place?

RogueMaster, 2023-04-16:
I will alert the original dev, additionally, you can try to explain the issue [over here](https://github.com/DarkFlippers/unleashed-firmware/)... they aren't the original dev but have a lot of SubGHz know how...

closed 2023-04-16 by RogueMaster

RogueMaster, 2023-04-25:
Is this still an issue @Geofferey ?