RogueMaster/flipperzero-firmware-wPlugins #612: bluetooth remote leads to NULL pointer dereference crash.

slikkmulah, 2023-04-29:
After factory reset > update 82.3 > RogueMaster 4/27 update > turning bluetooth on.

Whenever I try to open the bluetooth remote app I am met with my flipper crashing and getting the "NULL pointer dereference" error.

[APP] qFlipper version 1.3.0 commit 01fcfec9 2023-03-27T02:51:38
[APP] OS info: Windows 10 Version 22H2 10 10.0.19045 Qt 6.4.2
[RPC] Starting RPC session...
[RPC] RPC session started successfully.
[RPC] (1) System Protobuf Version START
[RPC] (1) System Protobuf Version SUCCESS
[RPC] (2) Property Get START
[RPC] (2) Property Get SUCCESS
[RPC] (3) Storage Info @/ext START
[RPC] (3) Storage Info @/ext SUCCESS
[RPC] (4) Storage Stat @/ext/Manifest START
[RPC] (4) Storage Stat @/ext/Manifest SUCCESS
[RPC] (5) System Get DateTime START
[RPC] (5) System Get DateTime SUCCESS
[RPC] (6) System Set DateTime START
[RPC] (6) System Set DateTime SUCCESS
[RPC] Stopping RPC session...
[RPC] RPC session stopped successfully.
[RPC] Starting RPC session...
[RPC] RPC session started successfully.
[RPC] (1) Gui Start ScreenStream START
[RPC] (1) Gui Start ScreenStream SUCCESS
[RPC] (2) Gui Send Input START
[RPC] (2) Gui Send Input SUCCESS
[RPC] (3) Gui Send Input START
[RPC] (3) Gui Send Input SUCCESS
[RPC] (4) Gui Send Input START
[RPC] (4) Gui Send Input SUCCESS
[RPC] (5) Gui Send Input START
[RPC] (5) Gui Send Input SUCCESS
[RPC] (6) Gui Send Input START
[RPC] (6) Gui Send Input SUCCESS
[RPC] (7) Gui Send Input START
[RPC] (7) Gui Send Input SUCCESS
[RPC] (8) Gui Send Input START
[RPC] (8) Gui Send Input SUCCESS
[RPC] (9) Gui Send Input START
[RPC] (9) Gui Send Input SUCCESS
[RPC] (10) Gui Send Input START
[RPC] (10) Gui Send Input SUCCESS
[RPC] (11) Gui Send Input START
[RPC] (11) Gui Send Input SUCCESS
[RPC] (12) Gui Send Input START
[RPC] (12) Gui Send Input SUCCESS
[RPC] (13) Gui Send Input START
[RPC] (13) Gui Send Input SUCCESS
[RPC] (14) Gui Send Input START
[RPC] (14) Gui Send Input SUCCESS
[RPC] (15) Gui Send Input START
[RPC] (15) Gui Send Input SUCCESS
[RPC] (16) Gui Send Input START
[RPC] (16) Gui Send Input SUCCESS
[RPC] (17) Gui Send Input START
[RPC] (17) Gui Send Input SUCCESS
[RPC] (18) Gui Send Input START
[RPC] (18) Gui Send Input SUCCESS
[RPC] (19) Gui Send Input START
[RPC] (19) Gui Send Input SUCCESS
[RPC] (20) Gui Send Input START
[RPC] (20) Gui Send Input SUCCESS
[RPC] (21) Gui Send Input START
[RPC] (21) Gui Send Input SUCCESS
[RPC] (22) Gui Send Input START
[RPC] (22) Gui Send Input SUCCESS
[RPC] (23) Gui Send Input START
[RPC] (23) Gui Send Input SUCCESS
[RPC] (24) Gui Send Input START
[RPC] (24) Gui Send Input SUCCESS
[RPC] (25) Gui Send Input START
[RPC] (25) Gui Send Input SUCCESS
[RPC] (26) Gui Send Input START
[RPC] (26) Gui Send Input SUCCESS
[RPC] (27) Gui Send Input START
[RPC] (27) Gui Send Input SUCCESS
[RPC] (28) Gui Send Input START
[RPC] (28) Gui Send Input SUCCESS
[RPC] (29) Gui Send Input START
[RPC] (29) Gui Send Input SUCCESS
[RPC] (30) Gui Send Input START
[RPC] (30) Gui Send Input SUCCESS
[RPC] (31) Gui Send Input START
[RPC] (31) Gui Send Input SUCCESS
[RPC] Serial connection was lost.
[RPC] Stopping RPC session...
[RPC] RPC session stopped successfully.
[RPC] Starting RPC session...
[RPC] RPC session started successfully.
[RPC] (1) System Protobuf Version START
[RPC] (1) System Protobuf Version SUCCESS
[RPC] (2) Property Get START
[RPC] (2) Property Get SUCCESS
[RPC] (3) Storage Info @/ext START
[RPC] (3) Storage Info @/ext SUCCESS
[RPC] (4) Storage Stat @/ext/Manifest START
[RPC] (4) Storage Stat @/ext/Manifest SUCCESS
[RPC] (5) System Get DateTime START
[RPC] (5) System Get DateTime SUCCESS
[RPC] (6) System Set DateTime START
[RPC] (6) System Set DateTime SUCCESS
[RPC] Stopping RPC session...
[RPC] RPC session stopped successfully.
[RPC] Starting RPC session...
[RPC] RPC session started successfully.
[RPC] (1) Gui Start ScreenStream START
[RPC] (1) Gui Start ScreenStream SUCCESS
[RPC] (2) Gui Send Input START
[RPC] (2) Gui Send Input SUCCESS
[RPC] (3) Gui Send Input START
[RPC] (3) Gui Send Input SUCCESS
[RPC] (4) Gui Send Input START
[RPC] (4) Gui Send Input SUCCESS

alecbramham, 2023-04-30:
Having the same issue as well. Looks like it can't find this file (/ext/apps_data/RemoteBT/.bt_hid.keys). I might see if I can fix this issue by finding that file in the standard firmware and copying it over.

631829 [I][SavedStruct] Loading "/ext/.config/desktop.settings"
633504 [I][LoaderSrv] Starting: Applications
633510 [I][AnimationManager] Unload animation 'Sasquach_D1g1talRa1n_128x64'
633528 [D][BrowserWorker] Start
633539 [D][BrowserWorker] Enter folder: /ext/apps items: 14 idx: -1
633542 [D][BrowserWorker] Load offset: 0 cnt: 50
634668 [D][BrowserWorker] Enter folder: /ext/apps/Bluetooth items: 3 idx: -1
634671 [D][BrowserWorker] Load offset: 0 cnt: 50
636214 [D][BrowserWorker] End
636226 [I][FapLoader] FAP Loader is loading /ext/apps/Bluetooth/RemoteBT.fap
636299 [I][FapLoader] FAP Loader is mapping
637295 [I][elf] Total size of loaded sections: 15899
637298 [I][FapLoader] Loaded in 1072ms
637300 [I][FapLoader] FAP Loader is starting app
637311 [I][BtGap] Stop advertising
637314 [D][BtGap] set_non_discoverable success
637532 [I][SavedStruct] Loading "/ext/.config/bt.settings"
637543 [E][SavedStruct] Failed to read "/ext/apps_data/RemoteBT/.bt_hid.keys". Error: file/dir not exist
637547 [E][BtKeyStorage] Failed to read payload size
637549 [I][FuriHalBt] Disconnect and stop advertising
637551 [I][FuriHalBt] Stop current profile services
637561 [I][FuriHalBt] Stop BLE related RTOS threads
637586 [I][FuriHalBt] Reset SHCI
637700 [I][FuriHalBt] Start BT initialization
637705 [I][Core2] Core2 started
637709 [I][Core2] C2 boot completed, mode: Stack
637712 [I][Core2] Core2: FUS: 1.2.0, mem 16/0, flash 6 pages
637714 [I][Core2] Core2: Stack: 1.15.0, branch 0, reltype 3, stacktype 3, flash 31 pages
637718 [I][Core2] Radio stack started

slikkmulah, 2023-05-01:
^ that would be awesome if you could. I dont know anout you guys but I actually use the BT remote semi frequently. So i been switching between rogue master and unleashed, which is pretty wack.

Definitely let me know if you find a solution. Much appreciated 🙏

Shocktrooper, 2023-05-01:
I am also here to report the same exact error.  Also not sure if its a bug or just the way it normally acts but when you choose BT Serial the screen just goes blank.  I assume this might be from the app waiting for a PC connection but it would be handy to let the end user know this is what this application needs

RogueMaster, 2023-05-04:
Trying to figure out what from changes in this branch makes it work again... https://github.com/RogueMaster/flipperzero-firmware-wPlugins/tree/WtfBT if any of you can try to find it, let me know if you find something 😄

lordrak007, 2023-05-04:
Hi, i just compiled your commit FAIL BETA 30 but unfortunatelly FZ still crashing. When i enter to BT/Remotes, i see list of remoted for 1 second and then FZ crashed.
How i can help to figure it out?
Lordrak

RogueMaster, 2023-05-04:
Thank you, you are correct. Troubleshooting further to figure out how I had it working yesterday with that build and now it no longer works, again. Back to square one...

RogueMaster, 2023-05-04:
Now testing with https://github.com/RogueMaster/flipperzero-firmware-wPlugins/commit/f98b8aba17bc6de765e0263b76ee0b28844d3d80

RogueMaster, 2023-05-05:
stack size on desktop is all that needed to be changed to fix this issue, next release will be fixed

closed 2023-05-05 by RogueMaster

slikkmulah, 2023-05-05:
> stack size on desktop is all that needed to be changed to fix this issue, next release will be fixed

Thanks dude, you da man.