RogueMaster/flipperzero-firmware-wPlugins #622: API Mismatch and BadUSB/Bad BT not working

grugnoymeme, 2023-05-08:
### Describe the bug.

Hello i built just now with ./fbt the fw but when i oush right arrow to see the passport, it result me API mismatch bla bla u want to continue?
and when i try open bad usb it returns "ERROR Preload failed INVALID FILE"

### Reproduction

Hello i built just now with ./fbt the fw but when i oush right arrow to see the passport, it result me API mismatch bla bla u want to continue?
and when i try open bad usb it returns "ERROR Preload failed INVALID FILE"

### Target

Bad USB / Passport

### Anything else?

no..

grugnoymeme, 2023-05-08:
installing the 508 via updater is working fine i dont know why

grugnoymeme, 2023-05-09:
ok maybe the problem is that i used ./fbt flash_usb that's not working with rm right?

RogueMaster, 2023-05-09:
No, it doesn't install apps on any firmware. You installed incorrectly and did not follow the instructions in the Read Me

closed 2023-05-09 by RogueMaster