RogueMaster/flipperzero-firmware-wPlugins #636: Remote Flipper app android

Kzy07, 2023-05-17:
### Describe the bug.

Depuis les 3 dernières mise à jour, je ne peux plus contrôler flipper depuis l'application Android,
- sub-ghz ne fonctionne pas 
- rfid ne fonctionne pas 
- nfc fonctionne 

Since the last 3 updates, I can no longer control pinball from the Android app,
- sub-ghz not working
- rfid not working
- nfc works

### Reproduction

Android app
Archive
Sub-ghz
'My file'
Send
Not remote flipper to app android

### Target

Flipper

RogueMaster, 2023-05-17:
Can you post a video of what you are trying to do?

Can you pair your Android device to flipper?

What SubGHz file is not working? I have confirmed SubGHz sends working on Flipper.

What RFID file is not working? I can confirm RFID is still working on Flipper.

Kzy07, 2023-05-18:
https://www.mediafire.com/file/rgxn65yyg1gtaqa/IMG_0193.MOV/file

RogueMaster, 2023-05-18:
This is likely due to renamed SubGHz and RFID apps so the app doesn't know how to launch these files. This is an issue with the app, kind of. The firmware will still work to play the files, you just need to navigate the firmware from the phone instead of trying to use the file shortcut. I'm open to ideas on how to fix this. Renaming SubGHz or RFID back to original names would cause them to show cut off on the Wii Menu view...

RogueMaster, 2023-05-18:
Should be fixed with https://github.com/RogueMaster/flipperzero-firmware-wPlugins/commit/7506ec92059eae9f630b111be350bb636fd702ce if you can, compile and test. Thank you!

closed 2023-05-18 by RogueMaster

Kzy07, 2023-05-18:
Hello, I just compiled. the problem is solved except for ibutton. thank you for your good work

Kzy07, 2023-05-18:
i just compiled 1889ec1 ibutton not work