RogueMaster/flipperzero-firmware-wPlugins #637: Application ' Temp Sensors Reader '

Kzy07, 2023-05-17:
### Describe the bug.

Bonsoir, dans apps/GPIO Temp Sensors Reader ne fait pas fonctionner mon DHT11. Le 5v fonctionne, même en essayant divers entrées 2,3,4 .... Mon DHT11 fonctionne parfaitement sur arduino mega.

Good evening, in apps/GPIO Temp Sensors Reader does not work my DHT11. The 5v works, even trying various 2,3,4 inputs.... My DHT11 works perfectly on arduino mega.

### Reproduction

Apps
GPIO
Temp Sensors Reader
DHT11
Pin 2
Save

### Target

Flipper zero

RogueMaster, 2023-05-17:
Please submit issues for this app here: https://github.com/quen0n/unitemp-flipperzero

Kzy07, 2023-05-17:
Sorry thx