RogueMaster/flipperzero-firmware-wPlugins #640: Excess Desktop Settings Moved to CFW App

RogueMaster, 2023-05-18:
### Describe the enhancement you're suggesting.

Non-Stock Settings to a Desktop section in CFW (Including Game Mode)

### Anything else?

Make sure Dice and Clock view the new Game Mode settings.

ZekePolarisBSH, 2023-06-12:
Is there a reason the PIN settings and everything under Desktop in the Flipper's settings always reset at random?

RogueMaster, 2023-06-13:
> Is there a reason the PIN settings and everything under Desktop in the Flipper's settings always reset at random?

Welcome to Flipper Zero development. Get used to it. They keep changing things related to the settings this they do reset.

ZekePolarisBSH, 2023-06-13:
Weird this happens without updates but I can get used to it. I just think of it as another relog from GTA5 servers.

RogueMaster, 2023-07-07:
> Weird this happens without updates but I can get used to it. I just think of it as another relog from GTA5 servers.

Make sure you don't have a desktop settings file on your SD card settings folders... RM FW is using the internal location for desktop settings...

RogueMaster, 2023-07-14:
@ESurge has completed this Bounty. ESurge confirm your BTC address on Discord DM for transfer

closed 2023-07-14 by RogueMaster