RogueMaster/flipperzero-firmware-wPlugins #641: IR Remote 5V

RogueMaster, 2023-05-18:
### Describe the enhancement you're suggesting.

Turn on 5V if IR settings is set to external device

### Anything else?

Should this be a setting somehow?

protocolpolice, 2023-05-19:
will try to merge  TV-B-Gone with flipper zero [https://twitter.com/_ProtocolPolice/status/1658169304188084237?s=20](url)

phr34k0, 2023-06-12:
Yeah as OP stated. I had troubles with this firmware but not Unleashed. Was trying to get my IR Blaster to work which it did not -> https://www.tindie.com/products/tehrabbitt/flipper-zero-ir-blaster/ as there was no 'set IR to external device'

roguepullrequest, 2023-06-12:
> Yeah as OP stated. I had troubles with this firmware but not Unleashed. Was trying to get my IR Blaster to work which it did not -> https://www.tindie.com/products/tehrabbitt/flipper-zero-ir-blaster/ as there was no 'set IR to external device'

You need to enable debug in the device settings menu then if the option doesnt show up then reboot.

RogueMaster, 2023-07-07:
After some testing, if GPIO gets 5V turned on, then IR Remote gets opened, it works perfect. So this issue is fixed I think... same behavior in RM and Unleashed... Also tested while viewing IR signal on phone camera and testing response from TV...

RogueMaster, 2023-07-07:
Re-opened since although it is not broken, having it in the app would be nice...

RogueMaster, 2023-07-16:
Best case, this scene can be added for the app... [infrared_scene_debug_settings](https://github.com/DarkFlippers/unleashed-firmware/blob/dev/applications/main/infrared/scenes/infrared_scene_debug_settings.c)

Perhaps Hold Back takes us to a menu to either Exit or GPIO Settings or Back To Remote... 

This would allow to enable 5V within app instead of having to exit and go to GPIO first...

RogueMaster, 2023-07-16:
@Hong5489 or @d4ve10 could you help?

RogueMaster, 2023-09-12:
If you are a Patreon supporter and would value this functionality, please DM on Patreon and I will add your monthly subscription to the prize allotment. If you have specific stipulations on how you want it to behave, feel free to detail them in a comment here for review.

coolshrimp, 2023-11-03:
> Best case, this scene can be added for the app... [...]

has this been solved yet? if not i can add it.

RogueMaster, 2023-11-03:
> has this been solved yet? if not i can add it.
It has not yet, please do :)

closed 2024-01-28 by RogueMaster