RogueMaster/flipperzero-firmware-wPlugins #652: Bluetooth not working since FW > RM0518-2052

stultitiophobia, 2023-06-02:
### Describe the bug.

After installing ANY release > RM0518-2052 bluetooth stops working.
I already forgot all devices in the phone and the flipper, but the phone does not find the flipper again.
After reinstalling FW RM0518-2052 it's working as expected again.

Does anyone else have this problem or does anyone have any idea ?

### Reproduction

Switch on
install ANY release > RM0518-2052
try bluetooth

RogueMaster, 2023-06-03:
How are you connecting? iOS or Android

RogueMaster, 2023-06-03:
Can confirm BT is working, try forgetting all devices and repair.

closed 2023-06-03 by RogueMaster

stultitiophobia, 2023-06-03:
> How are you connecting? iOS or Android

Hi, sorry - totally forgotten: iOS latest version.

stultitiophobia, 2023-06-12:
Just as feedback:
all working again, deleted all BT devices and reconnected, thanks a lot 👍🏻