RogueMaster/flipperzero-firmware-wPlugins #654: Setting favorite apps in settings has a bug

Mr-Rixa, 2023-06-04:
### Describe the bug.

Setting favorite apps in desktop settings, selecting "Applications" has previously opened app list to select an app to set as favorite, now it sets the application list as the favorite app

### Reproduction

1. Switch on
2. Navigate to "Settings"
3. Select "Desktop"
4. Select "x Favorite App"
5. Select "Applications"

Now on FW RM0523-2142-0.83.2-53a33f6 this would open Application list and you'd select your app, but on FW RM0602-0936-0.83.2-8927831 this sets the Applications list as the favorite app

### Target

RM0602-0936-0.83.2-8927831

### Logs

```Text
N/A
```

ESurge, 2023-06-05:
While not in a release yet, the fix for this has been pushed to source code.

The change is when selecting a Favorite, go down to the option: `[External Application]`. When selecting this option, it will open the file browser to select an external app. I may reconfigure it so it'll be like the old way of selecting `Application` and it opens the file browser, but for now this is the way.

ESurge, 2023-06-05:
I rewrote the fix and submitted it to be merged into the code. The rewrite makes it act like it used to. Selecting `Applications` will bring up the file browser as it once did before.

closed 2023-06-05 by RogueMaster