RogueMaster/flipperzero-firmware-wPlugins #663: Might be a bug I am not sure

ZekePolarisBSH, 2023-06-12:
### Describe the bug.

I am trying to use MIDI but it never loads up. I am trying to use it on this one opensource MIDI thing. Not sure if it is that thing, or it is just this app isn't working right now. Excuse my method of using GitHub's bug reporting since bug reports never made any sense to me... And why does ALL apps say mismatched API but for the most part works just fine? Is this just a thing that happens all the time? Just wondering

### Reproduction

Starting MIDI in USB Apps?

RogueMaster, 2023-06-13:
Please open a ticket with the MIDI app creator. Be sure to include more information about your MIDI thing.

RogueMaster, 2023-06-13:
For mismatched API delete your apps folder and reinstall RM

closed 2023-06-13 by RogueMaster

ZekePolarisBSH, 2023-06-13:
It is reinstalled though but this happens with official version for me as well so it could be just a common thing. And will do at MIDI app creator. I am reclosing this I just wanted to reply is all.

ZekePolarisBSH, 2023-06-13:
Wait a minute... Who is the app creator? It says you created it on the site. RIP