RogueMaster/flipperzero-firmware-wPlugins #675: Something happened

ZekePolarisBSH, 2023-06-25:
### Describe the bug.

Was working just fine and then out of no where it reverted back to dev but claims to be RM and it isn't letting me run any updates and all the tools are gone. I didn't do anything for this to happen. Any ideas?

### Reproduction

Normal use

RogueMaster, 2023-06-25:
Sounds like you are in game mode... What happens exactly when you press up and when you press left?

closed 2023-06-25 by RogueMaster

ZekePolarisBSH, 2023-06-29:
> Sounds like you are in game mode... What happens exactly when you press up and when you press left?

No that is not the case. It deleted everything including the folder that holds all that is RogueMaster. Gamemode is off.

ZekePolarisBSH, 2023-06-29:
Nothing happens when I press left btw.