RogueMaster/flipperzero-firmware-wPlugins #683: double wifi-mapping

grugnoymeme, 2023-06-28:
### Describe the enhancement you're suggesting.

There are two apps for esp32 wifi mapping, and also under apps_data there are two folders containing datas of the mapper.. can't understand why.. im sorry.. many thanks...
can i ask u also which of the tama_p1 folder is to keep? the one in the root of the SD containing the file "save.bin" or the one under apps_data containing the file "rom.bin", or both of them? thank u again.

grugnoymeme, 2023-06-28:
also in apps/settings directory there are two "Passport" with the same icon, but one of those is the passport settings.. it miss "settings"

grugnoymeme, 2023-06-28:
and there are also 2 "CF" and "CF SETTINGS" under apps/setting.
and the scroll stop when you reach the bottom, but it go infinite up.
sorry i just realizing it now.
I don't want to be annoying.
thank you very much

RogueMaster, 2023-06-28:
Two CFW settings listed under Settings is a known consequence of double linking the app as we are. Disregard. Aside from that, delete your entire apps folder to remove dupe apps and reinstall RM firmware. Also, remove tama folder in the root directory, RM tama uses the tama folder in the apps_data directory

closed 2023-06-28 by RogueMaster