RogueMaster/flipperzero-firmware-wPlugins #696: Air Tag for FZ ($50 USD in ETH or BTC)

RogueMaster, 2023-07-07:
### Describe the enhancement you're suggesting.

`Bluetooth OpenHayStack` bluetooth mode, only activates with special cloned key file present. Allows to operate like AirTag. Original Air Tag needs to be scanned for ID and then code cloned to a file on SD.

Bluetooth Settings needs a toggle change to be Off, On & OHS.
Internal Bluetooth needs to be updated to be on and transmit the ID if it it is set to OHS...
The ID transmitted needs to be opened from a txt file saved on the SD card so the user can manually change it.

[Feature explained here](https://github.com/flipperdevices/flipperzero-firmware/issues/1037)
[Past projects by AlexStrNik](https://github.com/AlexStrNik/flipperzero-firmware)
[seemoo-lab OHS](https://github.com/seemoo-lab/openhaystack)

Dismissal by OFW is invalid as it was made [here](https://github.com/culturally/flipper-zero-airtag) but hardcoded with someone's Air Tag.

### Anything else?

($50 USD in ETH or BTC) PR Submission and DM on Discord with wallet address.

RogueMaster, 2023-09-12:
If you are a Patreon supporter and would value this functionality, please DM on Patreon and I will add your monthly subscription to the prize allotment. If you have specific stipulations on how you want it to behave, feel free to detail them in a comment here for review.

RogueMaster, 2024-01-31:
Seems like [a BT OHS solution](https://github.com/flipperdevices/flipperzero-firmware/pull/3408) has been found, we need it to work with [the BLE profile rework](https://github.com/flipperdevices/flipperzero-firmware/pull/3272) for it to stick around... Thank you @thejudge156

UberGuidoZ, 2024-03-05:
https://www.reddit.com/r/flipperzero/s/0hMWqcjwd5

https://github.com/MatthewKuKanich/FindMyFlipper

MatthewKuKanich, 2024-03-08:
Release today!

UberGuidoZ, 2024-03-10:
@RogueMaster Looking at what Matthew released, it seems to fill this perfectly.

RogueMaster, 2024-03-10:
I agree!

closed 2024-03-11 by RogueMaster