RogueMaster/flipperzero-firmware-wPlugins #701: Buttons do not work when using RM firmware

Jdjdis, 2023-07-08:
### Describe the bug.

When you go install RougeMaster without animations, you are unable to use the buttons, and makes the flipper unusable 

### Reproduction

1. Install rouge master without animations

RogueMaster, 2023-07-08:
There may have been an issue with your installation, please install again.

closed 2023-07-08 by RogueMaster

Jdjdis, 2023-07-08:
I have reinstalled 5 times, and same issue

RogueMaster, 2023-07-08:
Try a different installation method maybe? How are you installing? I have installed the GitHub release No Animations and the Patreon release 2 times each and both are working upon install.

Jdjdis, 2023-07-08:
Installing on GitHub, via web installer. I also just installed using .fap
file and same error.

RogueMaster, 2023-07-08:
FAP files don't install the firmware... FAP files should just be used out of the firmware they are made for... also... web installer on GitHub for No Animations build is fully working and tested three times now...