RogueMaster/flipperzero-firmware-wPlugins #702: Added static frequency’s don’t appear in weather station app.

Jdjdis, 2023-07-08:
### Describe the bug.

When you go into custom firmware settings, you can add static frequencies.
When you go into the weather station app, these freauencv's do not reflect in the app.

### Reproduction

1. Go to CFW Settings
2. Click on protocols
3. Click on static frequencies
4. Add frequency
5. Exit out of CFW settings
6. Open applications app
7. Scroll to Sub-GHz
8. Click Sub-GHz
9. Scroll to Weather station
10. Click weather station
11. Click the left arrow to access weather station settings
12. Scroll through all frequencies, but the frequency you added will not appear

RogueMaster, 2023-07-08:
The frequency adding is not for weather station. Feel free to request this from OFW to be able to add frequencies to weather station.

closed 2023-07-08 by RogueMaster

RogueMaster, 2023-07-10:
Nevermind, found where to fix and will be fixed for next release

Jdjdis, 2023-07-11:
THANK YOU SO MUCH!!!!!!!!!!