RogueMaster/flipperzero-firmware-wPlugins #703: Flipper crashed and was rebooted (frui_check failed)

cokyrain, 2023-07-12:
### Describe the bug.

Use flipper app connected to flipper , enter a sub-ghz in archive and press send ,then the flipper crashed

### Reproduction

Use flipper app on ios, and firmware compiled from commits on Jul 11 and then on,

cokyrain, 2023-07-12:
furi_check failed

RogueMaster, 2023-07-12:
So this issue only happens when you are accessing through the app?

cokyrain, 2023-07-12:
> 
yes, it's only crash through the app. Direct operation on the machine without failure

closed 2023-07-12 by RogueMaster

cokyrain, 2023-07-13:
It's crashed at send sub-ghz when connected to mobilephone through bluetooth every time

cokyrain, 2023-07-13:
> This will be reviewed further

After testing, It's crashed through the app to send sub-ghz, and causes the file 'SDCard/subghz/assets/setting_user.txt’ to be emptied. Then after the flipper reboot , it's crashed everytime when send sub-ghz without mobilephone connected.